intel/mavlink-vehicles #17: Install mavlink

guiccbr, 2016-09-26:
We're not currently installing the mavlink submodule. This is needed in order to generate the mavlink messages properly. We should either install it in our own cmake or put Mavlink as a dependency on the README.md. I'm inclined to the first option.

mbelluzzo, 2016-09-26:
If you are going to keep it like a submodule and compile as part of our building process, I think you shouldn't install it. It can mess with the user previous installation (it would do in my machine for example). If it is to be a submodule, keep the usage like ardupilot does (totally internal). If it is to be a dependency like any other library, I do prefer to remove the submodule and add to the dependency list on Readme.md and the user needs to compile/install himself.

guiccbr, 2016-09-27:
Got your point. I'll try to keep it as Ardupilot then.

PR #20 (2016-09-29): Remove mavlink submodule
Closes #17 

Mavlink becomes a dependency instead.
closed 2016-09-29 by guiccbr