intel/mavlink-vehicles #43: mavlink bundling vs installed

lucasdemarchi, 2016-12-12:
On several systems it's not expected to have mavlink libraries installed. Particularly because of the way it's structured, having different code generated depending on the flight stack it's used. Another approach is to bundle the header-only library. This is the approach used by ardupilot, mavlink-router and PX4 (although PX4 pre-generates it, probably due to the GPL requirement).

See the approach in mavlink-router: https://github.com/01org/mavlink-router/blob/master/Makefile.am#L22-L27

guiccbr, 2016-12-13:
Thanks, @lucasdemarchi. We were actually using that other approach some time ago, as you can see in [here](https://github.com/01org/mavlink-vehicles/blob/94c50012ec6d5dbd360f23354721c4f8138a6445/CMakeLists.txt#L10). Can you please take a look at it and see if that's ok before we roll back?

mbelluzzo, 2017-02-09:
It doesnt need to be "installed", you just have provide a path to where it is.

mbelluzzo, 2017-02-16:
#47

closed 2017-02-17 by mbelluzzo